olikraus/u8glib #468: how to specify that a font is monospaced?

waspinator, 2017-08-18:
I'm trying to generate a 36 point monospaced font (https://fonts.google.com/specimen/Roboto+Mono), but it doesn't behave properly on my LCD.

I converted `RobotoMono-Regular.ttf` to a `bdf` font using font forge, and then I converted the `bdf` font to a `ug8` font using the `bdf2ug8.exe` utility. 

These are the important parts of my sketch. One using a built in monospaced font (profont29), and one using my generated version (roboto36).

```
sprintf(g_output_string, "|%4d|", my_number);

void draw(void) 
{
  u8g.setFont(u8g_font_profont29);
  u8g.setFontPosTop();
  u8g.drawStr(20, 10, g_output_string);
}

void draw(void) 
{
  u8g.setFont(roboto36);
  u8g.setFontPosTop();
  u8g.drawStr(20, 10, g_output_string);
}

```
for the `my_number` = `4` I get these outputs:

- `u8g_font_profont29`: `|   4|` (there are additional spaces, just not showing up here for some reason) 
- `roboto36`: `|4|`

```
const u8g_fntpgm_uint8_t roboto36[7415] U8G_FONT_SECTION("roboto36") = {
  0,21,41,0,246,26,6,251,15,164,32,255,248,31,246,30,
  248,255,4,26,26,22,9,0,224,224,224,224,224,224,224,224,
  224,224,224,224,224,224,224,224,224,224,0,0,0,0,224,240,
  240,224,9,9,18,22,6,19,225,128,225,128,225,128,225,128,
  225,128,225,128,225,128,225,128,225,128,20,26,78,22,1,0,
  0,193,192,0,193,128,1,195,128,1,195,128,1,131,128,1,
  131,0,1,131,0,127,255,240,127,255,240,127,255,240,3,6,
  0,3,6,0,7,6,0,7,14,0,7,14,0,6,14,0,
  255,255,192,255,255,192,255,255,192,14,28,0,12,28,0,12,
  24,0,12,24,0,28,56,0,28,56,0,24,56,0,16,35,
  70,22,3,252,1,192,1,192,1,192,1,192,7,224,31,248,
  63,252,60,62,120,14,112,15,112,15,112,7,112,7,120,0,
  120,0,62,0,31,128,15,240,7,248,0,252,0,62,0,15,
  0,15,0,7,224,7,224,7,240,7,112,15,124,30,63,252,
  31,248,7,224,1,128,1,128,1,128,20,26,78,22,1,0,
  62,0,0,127,0,0,255,128,0,227,128,0,193,131,0,193,
  134,0,193,142,0,227,140,0,255,156,0,127,24,0,28,56,
  0,0,112,0,0,96,0,0,224,0,0,199,128,1,207,192,
  1,159,224,3,152,112,7,56,112,6,56,112,14,56,112,4,
  56,112,0,24,112,0,31,224,0,15,224,0,7,128,19,26,
  78,22,2,0,3,224,0,15,240,0,31,248,0,28,60,0,
  60,28,0,56,28,0,56,28,0,60,56,0,28,120,0,30,
  240,0,15,224,0,15,192,0,15,128,0,31,192,0,61,193,
  192,121,225,192,240,241,192,240,121,192,224,59,128,224,63,128,
  224,31,0,240,15,0,120,31,128,127,255,128,63,251,192,15,
  193,224,2,9,9,22,9,19,192,192,192,192,192,192,192,192,
  192,10,37,74,22,6,248,0,128,1,128,3,192,7,0,14,
  0,14,0,28,0,28,0,56,0,56,0,120,0,112,0,112,
  0,112,0,112,0,240,0,240,0,240,0,240,0,240,0,240,
  0,240,0,112,0,112,0,112,0,112,0,112,0,56,0,56,
  0,56,0,28,0,28,0,14,0,7,0,3,128,1,192,0,
  128,9,37,74,22,6,248,128,0,192,0,224,0,112,0,56,
  0,56,0,28,0,30,0,14,0,14,0,7,0,7,0,7,
  0,7,0,7,128,3,128,3,128,3,128,3,128,3,128,3,
  128,3,128,7,128,7,128,7,0,7,0,7,0,15,0,14,
  0,14,0,28,0,28,0,56,0,112,0,112,0,224,0,128,
  0,17,18,54,22,3,4,1,192,0,1,192,0,1,192,0,
  1,192,0,1,192,0,65,193,0,241,143,0,255,191,128,127,
  255,0,7,240,0,3,192,0,7,224,0,14,112,0,14,120,
[... 401 lines cut ...]
  240,224,224,112,224,112,224,112,224,112,224,112,224,112,224,240,
  240,224,127,224,63,192,6,0,14,14,28,22,4,3,227,128,
  113,192,48,192,56,224,28,112,14,56,14,56,7,28,14,56,
  28,112,28,112,56,224,113,192,227,128,19,26,78,22,1,0,
  4,0,0,60,0,0,252,0,0,220,0,0,28,4,0,28,
  14,0,28,14,0,28,28,0,28,24,0,28,56,0,28,48,
  0,28,112,0,28,96,0,0,224,0,0,195,192,1,195,192,
  3,135,192,3,143,192,7,13,192,6,25,192,14,49,192,4,
  127,224,0,127,224,0,1,192,0,1,192,0,1,192,20,26,
  78,22,1,0,4,0,0,60,0,0,252,0,0,252,0,0,
  28,8,0,28,28,0,28,24,0,28,56,0,28,48,0,28,
  112,0,28,96,0,28,224,0,28,192,0,1,195,0,3,159,
  192,3,189,224,7,56,96,7,0,96,14,0,224,12,1,192,
  28,3,128,8,7,0,0,14,0,0,28,0,0,63,240,0,
  63,240,20,26,78,22,1,0,28,0,0,127,0,0,247,128,
  0,195,128,0,3,130,0,7,3,0,30,7,0,15,142,0,
  1,142,0,193,156,0,227,156,0,255,184,0,62,48,0,0,
  112,0,0,97,192,0,227,192,0,195,192,1,198,192,1,142,
  192,3,156,192,7,24,192,3,63,240,0,63,240,0,0,192,
  0,0,192,0,0,192,255,255,255,255,255,255,255,255,255,255,
  255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,
  255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,
  255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,
  255,255,255,255,255,255,255};
```

olikraus, 2017-08-18:
I do not remember exactly how my old converter works. Are you able to switch to u8g2? The bdf converter for u8g2 has a commandline option to force monospaced fonts. In fact i had worked a lot to improve this for u8g2.

waspinator, 2017-08-18:
I should have looked into u8g2 sooner. I didn't even have to create a 32 sized font, as it already had it. Thanks for all your hard work on these libraries.

closed 2017-08-18 by waspinator