olikraus/u8glib #469: EW50850FLWP - UC1611 driver - 8080 Mode (240W x 160H)

arnoldvillasanta, 2017-08-21:
Request to have this included in the supported LCD [Specs of LCD is here] if no constructor yet.
 (http://www.datasheet39.com/PDF/921659/EW50850FLWP-datasheet.html)

I wired the LCD IN 6080 mode (BM0 tied to 3.3V) and used the below constructors and pin connections but it didn't work:

- //U8GLIB_UC1611_DOGM240 u8g(8, 9, 10, 11, 4, 5, 6, 7, 18, 3, 17, 16); // 8Bit Com: D0..D7: 8,9,10,11,4,5,6,7 en=18, cs=3, di/a0=17,rw=16 
- //U8GLIB_UC1611_DOGXL240 u8g(8, 9, 10, 11, 4, 5, 6, 7, 18, 3, 17, 16); // 8Bit Com: D0..D7: 8,9,10,11,4,5,6,7 en=18, cs=3, di/a0=17, rw=16

en = 18 (WR1 of the LCD)
cs = 3 (CS1 of the LCD)
di/a0 = 17 (CD of the LCD)
rw = 16 (WR0 of the LCD) 
BM0 of LCD is in 3.3V to set to 8080 mode
RST of LCD is not connected

I tried to run Hello World sample code but nothing shows up.

olikraus, 2017-08-21:
let us continue with the  u8g2 ticket.

arnoldvillasanta, 2017-08-21:
thanks...  Noted regarding this one.