olikraus/u8glib #473: Problems with my oled display.

AceP996, 2017-10-22:
Recently I bought an oled display and am having some issues with it.
1 °: I do not know what his model, the seller did not specify in the advertisement ... he has 6 pins (VCC, GND, CLK, MOSI, CS and D / C) he is 128x64.
2 °: The image is being scrambled, not printing correctly on the display, and I do not know how to solve this problem.
If anyone can help me, I appreciate it.

olikraus, 2017-10-22:
Which constructor did you use to produce the pictures above?
Can provide a picture from a standard u8g2 example, e.g. HelloWorld

olikraus, 2017-10-22:
also, please consider to use u8g2. I am not working on u8glib any more.

AceP996, 2017-10-22:
I'm using the U8GLIB_SSD1306_128X64 u8g constructor (10, 9, 12, 11).
;

AceP996, 2017-10-22:
I'm using the U8GLIB_SSD1306_128X64 u8g constructor (10, 9, 12, 11);

olikraus, 2017-10-22:
maybe you should try the SH1106 constructor.

AceP996, 2017-10-22:
this constructor is not working in the compiler.
[image]

Gabapentin, 2017-10-22:
Hi, ACE 
We can see your entire sketch code to try?

AceP996, 2017-10-23:
the problem of the image being scrambled I managed to solve. but now a new problem has appeared ... when I use a delay, even if it is 0.1 seconds, the display is too slow to update or even update, for example, when I make a clock using the display, I have a lot of problems if I use the delay.
Code that i am using:

#include <U8glib.h>
#include <DS3231.h>
DS3231 rtc(SDA, SCL);
U8GLIB_SSD1306_128X64 u8g(10, 9, 12, 11);   
const int LM35 = A0; // Define o pino que lera a saída do LM35
float temperatura; // Variável que armazenará a temperatura medida

void mostrarHora(void){
  //funcoes do display pra mostrar horas
  u8g.setFont(u8g_font_osr21);
  //char *tempo = rtc.getTimeStr();
  //tempo[5] = '\0';
  u8g.drawStr(10, 57, rtc.getTimeStr());
  
  u8g.drawRFrame(0,18, 128, 46, 4);
}

/*
 void mostrarTemperatura(void){
  temperatura = (float(analogRead(LM35))*5/(1023))/0.01;
  //String temp = String(temperatr)
  //Serial.println(temperatura);
  String temp = String(temperatura);
  char *aux = temp[0];
  u8g.setFont(u8g_font_fub30);
  u8g.drawStr(10, 57, aux); 
 }

 */
 

void setup() {
    //configurações
   // Inicialização da comunicação
 Serial.begin(9600);
 // Initialização do rtc
  rtc.begin();
  rtc.setDOW(SUNDAY);     // Deve ser alterado o dia da semana para o dia em que estás em inglês
  rtc.setTime(15, 0, 0);     // Escreve as horas no formato 12:00:00 (formato 24 horas)
  rtc.setDate(22, 10, 2017);   // Escreve a data de acordo com o sistema que utilizas DD, MM, AAAA
  // Este programa escreve a data e hora no RTC e o relógio fica a trabalhar
  if ( u8g.getMode() == U8G_MODE_R3G3B2 ) {
    u8g.setColorIndex(255);     // white
  }
  else if ( u8g.getMode() == U8G_MODE_GRAY2BIT ) {
    u8g.setColorIndex(3);         // max intensity
  }
  else if ( u8g.getMode() == U8G_MODE_BW ) {
    u8g.setColorIndex(1);         // pixel on
  }
  else if ( u8g.getMode() == U8G_MODE_HICOLOR ) {
    u8g.setHiColorByRGB(255,255,255);
  }

}

void loop() {
  u8g.firstPage();
  while(u8g.nextPage()){
      mostrarHora();
      delay(1000);
  }
}

Code1110, 2017-10-24:
You need to change your code to sth. like this:

```c
void mostrarHora(void){
	//funcoes do display pra mostrar horas
	char *tempo = rtc.getTimeStr();
	//tempo[5] = '\0';

	u8g.firstPage();
	while(u8g.nextPage())
	{
		u8g.setFont(u8g_font_osr21);
		u8g.drawStr(10, 57, tempo);
		u8g.drawRFrame(0,18, 128, 46, 4);
	}
}

void loop() {
	mostrarHora();
	delay(1000);
}
```

You had at least two issues:
* You are getting the time again and again before it has been written to the display.
* You had the delay inside of the diplay-writing loop.

vainicius, 2022-05-11:
COMO DEIXA AMARELO