olikraus/u8glib #474: u8g_InitComFn with pin list (U8glib-ARM, Marlin 32 bit)?

Bob-the-Kuhn, 2017-10-29:
Don't spend more than 15 minutes on this.  I have a solution but would like an easier to duplicate method.  

U8G2 has a nice way of connecting a U8G2 device to a custom com function.    Unfortunately U8G2 costs more FLASH than U8G so I'm stuck with U8G.

Is there a way of using **u8g_InitComFn** with a pin list?  This would be using the U8glib-ARM library.

All the **u8g_InitComFn** examples I've found have the pins hard coded into the custom com function.  I need something where I can pass in the pins.

The goal is to be able to attach a custom com function to an existing U8G device with minimal code.  This is going to be repeated for multiple printers and multiple CPUs so easy to understand is highly desired.

I was hoping that something like this would work but so far no joy.
```
#include <U8glib.h>
uint8_t u8g_com_HAL_sw_spi_fn(u8g_t *u8g, uint8_t msg, uint8_t arg_val, void *arg_ptr);
uint8_t temp = u8g_InitComFn(&u8g, &u8g_dev_st7565_64128n_2x_sw_spi, u8g_com_HAL_sw_spi_fn);

class U8GLIB_64128N_2X_HAL : public U8GLIB 
{
  public:
    U8GLIB_64128N_2X_HAL(uint8_t sck, uint8_t mosi, uint8_t cs, uint8_t a0, uint8_t reset = U8G_PIN_NONE) 
      : U8GLIB(&u8g_dev_st7565_64128n_2x_sw_spi, sck, mosi, cs, a0, reset)
      { }
};

U8GLIB_64128N_2X_HAL u8g(DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0);
.
.
.
drawing routines
```

I have been able to copy existing device files, make some minor changes and include my custom com function there.  The new device is called from the main function via a new class.  Attached is one example.  I was hoping there is an easier to follow method.
[U8G_custom_com.zip](https://github.com/olikraus/u8glib/files/1424830/U8G_custom_com.zip)

olikraus, 2017-10-29:
I do not much remember the old u8glib code. Flexible pin handling was only there with the Arduino variant. Going back to the old code would probably cost much more than 15min :-(
 
What is the reason why u8g2 requires more flash? Any measurements? What is the difference in bytes?
One known problem is Wire.h. It is linked even if it is not used. To avoid this, you need to disable I2C in u8x8.h (u8g2 project, see FAQ).

Bob-the-Kuhn, 2017-10-29:
Thanks for taking the time to answer.  Much appreciated.

I have ARDUINO_AVR_DIGISPARK enabled which will disable U8X8_HAVE_HW_I2C.  Even with that U8G2 is about 1300-1400 bytes more than U8G.

olikraus, 2017-10-29:
hmmm.. I see...