olikraus/u8glib #475: Question

Manu8235, 2017-11-02:
Hallo together,

iam new here. I dont know if its the right side to ask a question.

I have now installed the u8glib, it works perfectly with the hallo world example, and want to create Bitmaps on AVR (Atmel Atmega328P. Screen st7565_dogm128_sw_spi)
How can I make this?
Same examples and a Bitmap creator will be very nice. 
or some links to help.

Thanks a lot.
Greezes

olikraus, 2017-11-02:
https://sandhansblog.wordpress.com/2017/04/16/interfacing-displaying-a-custom-graphic-on-an-0-96-i2c-oled/

Manu8235, 2017-11-03:
Thanks.
Which Funktion should I take?
In the example is u8g2.drawXBMP.
I tested u8g_DrawHBitmap.
like this:

void draw(void) {
	u8g_SetFont(&u8g, u8g_font_4x6);
	u8g_DrawHBitmap(42,0,Ironman_width,Ironman_height,Ironman_bits);
}
but its the old Screen.

Thanks for help.

olikraus, 2017-11-04:
Indeed, maybe you should better move to the new u8g2 library:
https://github.com/olikraus/u8g2

You can also install u8g2 via Arduino IDE library manager.

Manu8235, 2017-11-04:
Thank you, also for the hole work.
I have it work, on the u8glib.

But i will test also the new lib.