olikraus/u8glib #476: NHD-2.7-12864WDY3 driver update

conectia, 2017-12-03:
Hi,

The new NHD-2.7 128x64 oled display comes with the ssd1322 controller instead of the ssd1325.

I have been trying to replace it with your ssd1322 256x64 controller.

The display works, but I get scrambled output. The display shows one half with the output compressed and the other one with garbage.

I think it will be easy to make it work touching a few bits.

Any idea.

Thanks in advance.

olikraus, 2017-12-04:
Can you link the datasheet and/or the product description?

conectia, 2017-12-04:
The product page : http://www.newhavendisplay.com/nhd2712864wdy3-p-9542.html

And two images, the old working ssd1325 and the result shown with the ssd1322
[image]
[image]

Thanks

conectia, 2017-12-08:
Partially solved !

Replacing the driver:

U8G2_SSD1325_NHD_128X64_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ A1, /* dc=*/ A0);
to
U8G2_SSD1322_NHD_256X64_1_4W_HW_SPI u8g2(U8G2_R2, /* cs=*/ A1, /* dc=*/ A0);

Setting   u8g2.setFlipMode(1); I get the horizontal mirror.

Updating   U8X8_CAA(0x0a0, 0x014, 0x011),                /* remap */
to   U8X8_CAA(0x0a0, 0x004, 0x011),                /* remap */

I get a vertical mirror.
But still the image is compressed horizontally.

Any idea ?

Thanks in advance.
[image]

olikraus, 2017-12-08:
Is this the same display: https://github.com/olikraus/u8g2/issues/412 ?

Does this issue belong to u8g2?

conectia, 2017-12-08:
Yes, I'll update issue 412

closed 2017-12-08 by conectia