olikraus/u8glib #477: get width of 'int'

snapedumbledore, 2017-12-27:
I would like to know how can i get the width of 'int' type, not a string.
Thanks.

olikraus, 2017-12-27:
This is more a C/C++ question, not a u8g2 related question. Nevertheless, the bytewidth can be calculated with "sizeof(int)"

t3chguy, 2017-12-27:
Unless of course he means get the width of a numeric character in a font given the number as an int

snapedumbledore, 2017-12-27:
Emmmm.Thanks.
I want to calculate the 'x' width which is int。（int x = 0,and x will be increased or decreased）

olikraus, 2017-12-27:
There is no other option: You have to convert the number (int) to a string. During this conversion you have to decide whether to include a minus sign, leading zeros, an additional % sign and which base you want to use (10/decimal, 16 hexadecimal, etc). All these decision during the string conversions will change the size of the number. So, i guess it will be best to use "sprintf" with  proper conversion options and then apply the resulting string to getStrWidth().