olikraus/u8glib #479: 2.9" e-Paper 296x128 - Access full display coordinate range

MajicDesigns, 2018-02-03:
**Environment:** Arduino Uno, Windows 10, u8g2 library

**Problem:** Trying to draw a frame  on the outside of the display but it will not work beyond width of 248. Display width is 296. Attached are images of what is drawn for width 248, 250 and 296. Note that for W 250 the right side of the frame is incomplete. For W 296 some sort of truncation seems to occur.
Code used to generate the display is also attached. 
I see a similar effect when trying to display a string beyond column 248 - it wraps back to the left hand side of the display.
I am pretty sure this should work, so can you please advise if there is something I am missing in order for this to show correct output?

**Images:**
W 248
[image]

W 250
[image]

W DISPLAY_WIDTH
[image]

**Code:**
```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

// WaveShare 2.9 inch eInk/ePaper Display, enable 16 bit mode for this display!
const uint8_t CLK_PIN = 13;
const uint8_t DATA_PIN = 11;
const uint8_t CS_PIN = 10;
const uint8_t DC_PIN = 9;
const uint8_t RST_PIN = 8;
U8G2_IL3820_296X128_1_4W_SW_SPI u8g2(U8G2_R0, 13, 11, 10, 9, 8);

const uint16_t DISPLAY_WIDTH = 296;
const uint16_t DISPLAY_HEIGHT = 128;

void setup(void) 
{
  uint16_t x, y, d;
  
  u8g2.begin();
  u8g2.setContrast(255);
  u8g2.setFont(u8g2_font_helvR12_tr);  // 12 point Helvetica
  u8g2.firstPage();
  do 
  {
    x = 248;
    y = DISPLAY_HEIGHT;
    d = 2;
    u8g2.drawFrame(0, 0, x, y);
    u8g2.drawFrame(d, d, x-(d*2), y-(d*2));
  } while (u8g2.nextPage());
}

void loop(void) 
{
}
```

olikraus, 2018-02-03:
hmm... i guess this is posted to the wrong projected... should be here: https://github.com/olikraus/u8g2/issues

Anyhow, did you enable 16 bit mode?
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L159

MajicDesigns, 2018-02-03:
Sorry for posting in the wrong place.

Yes, enabling 16 bit mode is what I was missing. Thanks for the quick response and your support.

closed 2018-02-03 by MajicDesigns

olikraus, 2018-02-03:
:)