olikraus/u8glib #481: U8GLIB_T6963_240X64 not working with nano and Arduino IDE 1.8.5

SailWithChips, 2018-03-07:
Everything compiles fine but the program get stuck on setup and never enter to loop although without using u8g... Could anyone help with this?
`U8GLIB_T6963_240X64 u8g(8, 9, 10, 11, 4, 5, 6, 7, 14, 15, 17, 2, 16); // 8Bit Com: D0..D7: 8,9,10,11,4,5,6,7, cs=14, a0=15, wr=17, rd=2, reset=16

void setup(void) {
  Serial.begin(115200);
}

void loop(void) {  
  delay(1000);
    int contrast = random(0,255);
    Serial.println(contrast);       

}`

olikraus, 2018-03-07:
Will u8g2 work?

SailWithChips, 2018-03-07:
Have already tested and also get stuck, also tested with a Arduino MEGA 2560 and same problem

olikraus, 2018-03-07:
u8g2 has a different arguments, so pins need to be adjusted. Also the rd pin has to be connected differently.
There are also a couple of other things which need to be considered for this type of displays, but with this little information you gave it is hard to say something.

SailWithChips, 2018-03-07:
Thanks for your help Olikraus, I already changed the pins and connected ‘rd’ to +5V.
The program was not stucked, it was something related with my serial, sorry for that. Now have ‘U8g’ working again in my project also in ArduinoIDE 1.8.5 but with ‘U8g2’, (and connecting propperly) my display only show first rows (about 15). These 15 rows are seen ok but the rest of display is empty... I tried the examples and also my project with same behaviour. If you need more information, let me know... by the way I’ll continuenusing ‘U8g’... thanks a lot for your good job!

olikraus, 2018-03-07:
so U8glib works?

SailWithChips, 2018-03-07:
Yes!

olikraus, 2018-03-07:
ok... :-)