olikraus/u8glib #482: Watchdog

spa-sam, 2018-03-08:
Watchdog not work!
Atmega328p, Nokia 5110. If Watchdog is triggered then the microcontroller restarts and freeze (or enters the reboot cycle?!). It does not depend whether there is a bootloader or not.

The display of the Nokia 5110 with the library from Adafruit wdt works.

spa-sam, 2018-03-08:
Watchdog not work!
Atmega328p, Nokia 5110. If Watchdog is triggered then the microcontroller restarts and freeze (or enters the reboot cycle?!). It does not depend whether there is a bootloader or not.

The display of the Nokia 5110 with the library from Adafruit wdt works.

`#include "U8glib.h"
U8GLIB_PCD8544 u8g(19, 18, 16, 17, 15);  // CLK=19, DIN=18, CE=16, DC=17, RST=15

#include <avr/wdt.h>

void setup() {
wdt_enable(WDTO_8S);
wdt_reset();

 u8g.firstPage();
do {  
    u8g.setFont(u8g_font_6x12);
    u8g.setPrintPos(1,8);
    u8g.print("Start...");  
                    } while(u8g.nextPage());

delay(2000);

}

void loop() {

wdt_reset();

 u8g.firstPage();
do {  
    u8g.setFont(u8g_font_6x12);
    u8g.setPrintPos(1,8);
    u8g.print("ReStarting...");  
                    } while(u8g.nextPage());

                    
delay(9000);
}`

olikraus, 2018-03-08:
hmm... so what shell I do?

Dominiquechai, 2018-06-20:
Try like as :
void setup() {
  MCUSR = 0; 
  cli(); 
  wdt_reset(); 
  wdt_enable(WDTO_8S); 
  WDTCSR = (1<<WDCE) | (1<<WDE); 
  WDTCSR = (1<<WDE); 
  WDTCSR = (1<<WDIE); // Watchdog interrrupt enabled.
  MCUSR = 0;
  sei();
}
I don't know why wdt_enable() does not work normally.
But, above code work well.

porumatu, 2019-01-31:
Hello I have the same problem with u8g library. Code above didn't work, could you please solve this problem?

AnHardt, 2019-01-31:
Sorry was from memory.
The 4 bits are in the same register but not consecutive. -> same result.