olikraus/u8glib #484: How to add symbol like(micro,degree,ohm etc)?

eRony-232, 2018-07-12:
I found u8glib is a good library for OLED.
Specially for AVR C language!

I have two question arise during working on it.
1.How to clear OLED?
2.How to add  symbol like(micro,degree,ohm etc)?

Regrads,
eRony

olikraus, 2018-07-12:
I do not work on u8glib any more. Please use u8g2.
U8g2 also supports Unicode and UTF-8 which will simplify the use of symbols.

eRony-232, 2018-07-14:
Okay.

I have tried to port U8g2 lib with I2C protocol in ATmega328P.

Initial sequence:
u8g2_t u8g2;
...........................
u8g2_Setup_ssd1306_i2c_128x64_noname_2(&u8g2, U8G2_R0, u8x8_byte_sw_i2c, u8x8_avr_delay); 

u8g2_InitDisplay(&u8g2);
u8g2_SetPowerSave(&u8g2, 0);
	
	/* full buffer example, setup procedure ends in _f */
	u8g2_ClearBuffer(&u8g2);
	u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
	u8g2_DrawStr(&u8g2, 1, 18, "U8g2 on AVR");
	u8g2_SendBuffer(&u8g2);

Result is Failed!
How to fix it?

eRony-232, 2018-07-15:
The u8x8_avr_delay is from here "u8g2/sys/avr/atmega328/main.c"
 
`uint8_t u8x8_avr_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	uint8_t cycles;

	switch(msg)
	{
		case U8X8_MSG_DELAY_NANO:     // delay arg_int * 1 nano second
			// At 20Mhz, each cycle is 50ns, the call itself is slower.
			break;
		case U8X8_MSG_DELAY_100NANO:    // delay arg_int * 100 nano seconds
			// Approximate best case values...
#define CALL_CYCLES 26UL
#define CALC_CYCLES 4UL
#define RETURN_CYCLES 4UL
#define CYCLES_PER_LOOP 4UL

			cycles = (100UL * arg_int) / (P_CPU_NS * CYCLES_PER_LOOP);

			if(cycles > CALL_CYCLES + RETURN_CYCLES + CALC_CYCLES) 
				break;

			__asm__ __volatile__ (
			"1: sbiw %0,1" "\n\t" // 2 cycles
			"brne 1b" : "=w" (cycles) : "0" (cycles) // 2 cycles
			);
			break;
		case U8X8_MSG_DELAY_10MICRO:    // delay arg_int * 10 micro seconds
			for(int i=0 ; i < arg_int ; i++)
				_delay_us(10);
			break;
		case U8X8_MSG_DELAY_MILLI:      // delay arg_int * 1 milli second
			for(int i=0 ; i < arg_int ; i++)
				_delay_ms(1);
			break;
		default:
			return 0;
	}
	return 1;
}
`

I have tried also with u8g2_Setup_ssd1306_i2c_128x64_noname_f ();
But still its showing nothing in OLED!

eRony-232, 2018-07-15:
It's Really helpful if you give an example code for avr's i2c.

olikraus, 2018-07-15:
In your callback function. Do you also consider the following two cases?
```
case U8X8_MSG_GPIO_I2C_CLOCK:		// arg_int=0: Output low at I2C clock pin
      break;							// arg_int=1: Input dir with pullup high for I2C clock pin
    case U8X8_MSG_GPIO_I2C_DATA:			// arg_int=0: Output low at I2C data pin
      break;							// arg_int=1: Input dir with pullup high for I2C data pin
```

> It's Really helpful if you give an example code for avr's i2c.

I do not have a test setup for this.

eRony-232, 2018-07-16:
Well.
I have test setup please guide me.

`case U8X8_MSG_GPIO_I2C_CLOCK:		// arg_int=0: Output low at I2C clock pin
      break;							// arg_int=1: Input dir with pullup high for I2C clock pin
    case U8X8_MSG_GPIO_I2C_DATA:			// arg_int=0: Output low at I2C data pin
      break;	`

I'm not including this line!

Where does it need to include?
And it is software i2c right?
Where is i2c pin defined?

olikraus, 2018-07-16:
Please read this article: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

> Where does it need to include? 

In the GPIO and delay procedure.

> And it is software i2c right?

Yes

> Where is i2c pin defined?

You can use any GPIO pin in case of software I2C.

eRony-232, 2018-07-17:
I have already read it.

I have found followed line in u8*8.h and line no 294,295:

`#define U8X8_PIN_I2C_CLOCK 12	/* 1 = Input/high impedance, 0 = drive low */
#define U8X8_PIN_I2C_DATA 13	/* 1 = Input/high impedance, 0 = drive low */`

12 and 13 No pins are used for software i2c right?

But AVR MCU does not these types of pin declaration!

eRony-232, 2018-07-17:
I have already read it.

I have found followed line in u8*8.h and line no 294,295:

#define U8X8_PIN_I2C_CLOCK 12 /* 1 = Input/high impedance, 0 = drive low */ #define U8X8_PIN_I2C_DATA 13 /* 1 = Input/high impedance, 0 = drive low */

12 and 13 No pins are used for software i2c right?

But AVR MCU does not these types of pin declaration!

olikraus, 2018-07-17:
> 12 and 13 No pins are used for software i2c right?

No, 12 and 13 are the values of messages. You need to react on these messages and set the output level accordingly.