olikraus/u8glib #485: Drawing string without re-drawing all screen

zmechu, 2018-08-21:
Hello!

I have simple questiuon.  Got 128x64 OLED I2C display, which forks fine with u8glib.
But as I see, to update screen informations, I must execute:
u8g.firstPage();
  do {  
    draw();    
  } while( u8g.nextPage() ); 
where draw(); got all drawing procedures I need (strings, graphics, data etc), which is re-draw all every time.

But is there a way to re-daw ONLY selected part.
For example, I have:
u8g.setPrintPos(0, 20);u8g.print(" SIGNAL:");u8g.print(x); 
where string " SIGNAL:" didnt change at all, only "x" changes. It is not nessesary to draw " SIGNAL:" all the time.

Is there a way to draw " SIGNAL:" once at start and further draw only "x" data, look like:
void setup () {
[...]
u8g.setPrintPos(0, 20);u8g.print(" SIGNAL:");
[...]
}

void loop() {
[...]
u8g.setPrintPos(0, 40);u8g.print(x);
[...]
}

olikraus, 2018-08-21:
In u8g2 you can use full buffer mode to do a partial update.

olikraus, 2018-08-21:
No, the full buffer mode was newly introduced with U8g2.

zmechu, 2018-08-21:
OK! 
Thank you for great job :-)

closed 2018-08-21 by olikraus