olikraus/u8glib #486: Atmel studio 7 errors

TheNightWhisper, 2018-09-08:
Hi My friends
I'm trying to make IR remote control receiver using atmega8 u controller, using atmel studio 7,
I've created the main file main.c and included the library header file, but when compiling the project I've get
punch of errors and after googling i found some one recommend changing the main file name from main.c to main.cpp ,and re compile, after doing this really it solve many many issues (and i don't know why ? !),
**_but i stuck with two errors_**

**1- ID returned 1 Exit Status
2- recipe for target IRRemote.elf failed**

so please if any one can help to get the hex file
main.cpp
`
#define recvPin 0
#define outPutPin1 1
#define outPutOin2 2
#include <avr/io.h>
#include "IRremote.h"

int main(void)
{
    IRrecv irRecv(recvPin);
	decode_results results;
	irRecv.enableIRIn();
	DDRC=(1<<2|1<<1|0<<0);
    while (1) 
    {
		if(irRecv.decode(&results))
		{
		 switch(results.value)
		 {
				case 0xFD00FF:
					PORTC=(1<<1);
				break;
		  		
				  case 0xFD807F:
					PORTC=(1<<2);
				
				break;
		 
			}
		}
	}
}

`
IRremot.h
`
#ifndef IRremote_h
#define IRremote_h

//------------------------------------------------------------------------------
// The ISR header contains several useful macros the user may wish to use
//
#include "IRremoteInt.h"

//------------------------------------------------------------------------------
// Supported IR protocols
// Each protocol you include costs memory and, during decode, costs time
// Disable (set to 0) all the protocols you do not need/want!
//

#define DECODE_NEC           1
#define SEND_NEC             1

//------------------------------------------------------------------------------
// When sending a Pronto code we request to send either the "once" code
//                                                   or the "repeat" code
// If the code requested does not exist we can request to fallback on the
// other code (the one we did not explicitly request)
//
[... 236 lines cut ...]

# Add inputs and outputs from these tool invocations to the build variables 

# All Target
all: $(OUTPUT_FILE_PATH) $(ADDITIONAL_DEPENDENCIES)

$(OUTPUT_FILE_PATH): $(OBJS) $(USER_OBJS) $(OUTPUT_FILE_DEP) $(LIB_DEP) $(LINKER_SCRIPT_DEP)
	@echo Building target: $@
	@echo Invoking: AVR/GNU Linker : 5.4.0
	$(QUOTE)C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin\avr-gcc.exe$(QUOTE) -o$(OUTPUT_FILE_PATH_AS_ARGS) $(OBJS_AS_ARGS) $(USER_OBJS) $(LIBS) -Wl,-Map="IRRemote.map" -Wl,--start-group -Wl,-lm  -Wl,--end-group -Wl,--gc-sections -mmcu=atmega8a -B "C:\Program Files (x86)\Atmel\Studio\7.0\Packs\atmel\ATmega_DFP\1.2.209\gcc\dev\atmega8a"  
	@echo Finished building target: $@
	"C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin\avr-objcopy.exe" -O ihex -R .eeprom -R .fuse -R .lock -R .signature -R .user_signatures  "IRRemote.elf" "IRRemote.hex"
	"C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin\avr-objcopy.exe" -j .eeprom  --set-section-flags=.eeprom=alloc,load --change-section-lma .eeprom=0  --no-change-warnings -O ihex "IRRemote.elf" "IRRemote.eep" || exit 0
	"C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin\avr-objdump.exe" -h -S "IRRemote.elf" > "IRRemote.lss"
	"C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin\avr-objcopy.exe" -O srec -R .eeprom -R .fuse -R .lock -R .signature -R .user_signatures "IRRemote.elf" "IRRemote.srec"
	"C:\Program Files (x86)\Atmel\Studio\7.0\toolchain\avr8\avr8-gnu-toolchain\bin\avr-size.exe" "IRRemote.elf"
	
	

# Other Targets
clean:
	-$(RM) $(OBJS_AS_ARGS) $(EXECUTABLES)  
	-$(RM) $(C_DEPS_AS_ARGS)   
	rm -rf "IRRemote.elf" "IRRemote.a" "IRRemote.hex" "IRRemote.lss" "IRRemote.eep" "IRRemote.map" "IRRemote.srec" "IRRemote.usersignatures"
`

TheNightWhisper, 2018-09-08:
[image]
this  a snap shot of my project

olikraus, 2018-09-08:
I can support regarding u8g but unfortunately not with irremote.

TheNightWhisper, 2018-09-10:
pls, olikraus, i didn't understand what you mean

TheNightWhisper, 2018-09-11:
many thanks olikraus

TheNightWhisper, 2018-09-20:
>  This is the u8glib issue tracker. The purpose is to discuss topics related to u8glib.
> From what I see, your question is not at all related to u8glib.
so pls where is can ask  this question ??