olikraus/u8glib #488: u8g_DrawBox() with h=0 corrupts display

francescolavra, 2018-10-29:
If the u8g_DrawBox() function in csrc/u8g_rect.c is called with the `h` argument set to 0, it corrupts the display by drawing a box that covers the entire height of the screen. This is because the body in the while() loop in `u8g_draw_box()` is executed before checking the value of the `h` argument.
This can be fixed by replacing the `do {} while()` construct with a simple while() loop:
```
while( h != 0 )
{
  u8g_draw_hline(u8g, x, y, w);
  y++;    
  h--;
}
```

olikraus, 2018-10-29:
True, but Code size will be higher.

olikraus, 2018-10-29:
Well... I can add this...

AnHardt, 2018-10-29:
A box with height 0 is still a line. For that, at least one line is required. How about changing the condition to `>=`

francescolavra, 2018-10-29:
> A box with height 0 is still a line

Why? This wouldn't be consistent with the current implementation, where a line corresponds to a box with height 1, not 0. Also, `u8g_DrawBox()` doesn't draw anything when called with width 0.