olikraus/u8glib #491: Using twoST7920 displays on one Arduino using SPI

dgoadby, 2019-02-10:
I am using an ST7920 display in SPI mode with an Arduino UNO. It works well. But I really need two displays in my project. I could use two Arduinos but the data changes very slowly so one Arduino would easily do the job. 

My first idea is to create two UG8 objects and then a select line for each display with two display loops. Is that possible or is the memory going to be a constraint?

My second idea is to use only one UG8 object and make the display selection outside of the display loop. Clearly the draw() code would be different for each display. As the screen is redrawn each time then it seems a plausible idea.

Any comments?

olikraus, 2019-02-10:
Both ideas should work. The u8g2 FAQ also includes some hints.

dgoadby, 2019-02-15:
Thanks. I will probably try the second option first as the memory constraints are less. With an STM Blue Pill the first option would probably work ok. I will feedback my results.

closed 2019-02-15 by dgoadby