olikraus/u8glib #494: SH1106 or SSD1306 on SW SPI with Atmel Studio 7 working slowly

elektronek, 2019-03-12:
Hello,
im use Atmel Studio 7 and u8g library 1.18.1 (or older version - the same effect) with atmega32 and 16MHz xtal, Oled displays SH1106 or SSD1306 - the same function

#include "u8g.h"

u8g_t u8g;

initialize:
u8g_InitSPI(&u8g, &u8g_dev_sh1106_128x64_sw_spi, PN(1, 3), PN(1, 4), PN(1, 7), PN(1, 6), PN(1, 5)); 

void draw1(void)
{
	u8g_SetFont(&u8g, u8g_font_6x10);
	u8g_DrawStr(&u8g, 0, 10, "Hello World! 1");
	u8g_DrawStr(&u8g, 0, 20, "Hello World! 2");
	u8g_DrawStr(&u8g, 0, 30, "Hello World! 3");
	u8g_DrawStr(&u8g, 0, 40, "Hello World! 4");
	u8g_DrawStr(&u8g, 0, 50, "Hello World! 5");
	u8g_DrawStr(&u8g, 0, 60, "Hello World! 6");
}

void draw2(void)
{
	 u8g_SetFont(&u8g, u8g_font_6x10);
	 u8g_DrawStr(&u8g, 0, 10, "Nazdar svete! 1");
	 u8g_DrawStr(&u8g, 0, 20, "Nazdar svete! 2");
	 u8g_DrawStr(&u8g, 0, 30, "Nazdar svete! 3");
	 u8g_DrawStr(&u8g, 0, 40, "Nazdar svete! 4");
	 u8g_DrawStr(&u8g, 0, 50, "Nazdar svete! 5");
	 u8g_DrawStr(&u8g, 0, 60, "Nazdar svete! 6");
}

and loop in main() function

    while (1) 
    {
		u8g_FirstPage(&u8g);
		do
		{
			draw1();
		} while ( u8g_NextPage(&u8g) );
		_delay_ms(1000);

		u8g_FirstPage(&u8g);
		do
		{
			draw2();
		} while ( u8g_NextPage(&u8g) );
		_delay_ms(1000);
    }

... and drawing slowly, one draw is abt 0.7 sec if the same code compiled and loaded from Arduino IDE with U8glib library, working fine fast. 

What is missing in my code ?  Thank you very much

olikraus, 2019-03-12:
Software emulated SPI is always a little bit slow.

Please note that I do not support u8glib any more and also note, that u8glib was designed for Arduino. So in general you need to implement your own HAL for u8g (or u8g2)

U8g2 has some more hints on this topic: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

elektronek, 2019-03-13:
Hello, 
thank you for reply, im try use u8g2, result is better (fastest) from u8g, 
but my bitmap in PROGMEM is not showed correctly

with old library u8g - correctly viewed
u8g_DrawBitmapP(&u8g, 0, 24, 16, 21, my_bitmap);

now im try - not correctly viewed
u8g2_DrawXBMP(&u8g2, 0, 24, 16, 21, my_bitmap); 

and if try
u8g2_DrawBitmap(&u8g2, 0, 24, 16, 21, my_bitmap); 

view is correct, but data not loaded from PROGMEM
what is correct using DrawXBMP ? data format is must be converted ?

thank you for your help with this last problem :-)

olikraus, 2019-03-13:
Yes, DrawBitmapP is not supported any more. I wanted users to use XBM only.
You have the original bitmap? You could use gimp to generate the XBM format.

elektronek, 2019-03-13:
Hello,
Thank you, im converted image with LCD Assistant and with u8g displayed correctly. 

U8g2 showing byte mirrorly, i.e.
0x01 is X - - - - - - -
and in u8g
0x01 is - - - - - - - X

i have original bitmap, now im try convert with Gimp

elektronek, 2019-03-13:
Hello, thank you very much for your help,. now working fine :-)

olikraus, 2019-03-16:
:)