olikraus/u8glib #495: V1.19.9 Crashes during compile on marlin 1.1.9-bugfix

Xeliaa, 2019-04-08:
Error log below, have to revert to 1.18.0 in order to install on a mks gen l board
=============================
Build options changed, rebuilding all
C:\Users\saeba\OneDrive\Documents\Arduino\libraries\U8glib\src\clib\u8g_rect.c: In function 'u8g_DrawBox.constprop':

C:\Users\saeba\OneDrive\Documents\Arduino\libraries\U8glib\src\clib\u8g_rect.c:126:1: internal compiler error: Segmentation fault

 }

 ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Users\saeba\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/users/saeba/appdata/local/arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

olikraus, 2019-04-08:
strange...

swilkens, 2019-04-08:
I have no problems building Marlin 1.1.x-bugfix using Arduino IDE 1.8.9 and U8glib V1.19.1.

Xeliaa, 2019-04-08:
yes it is strange because everytime i update it to 1.19 it crashes and im running 1.8.9 ide as well, i always seem to find the bugs that cant be reproduced, my machines are cursed. i'll try a fresh install and see what happens.

closed 2019-04-09 by Xeliaa