olikraus/u8glib #496: Is it possible to use spi2 hw on atmega328pb?

tistructor, 2019-08-23:
I'm using an atmega328pb with arduino and I would like to use the spi2 to control an oled ssd1309 display, is it possible? or should I use the display in sw?
thank you

olikraus, 2019-08-23:
There is only one SPI available on your uC. You can use SW SPI or you can share the one HW SPI with other devices.

tistructor, 2019-08-24:
atmega328pb has 2 spi, 2 i2c, 2 uart

tistructor, 2019-08-24:
for the device I am building I cannot use the normal spi0 pins (10,11,12,13) we must use spi1 (19, 22, 23, 24) for problems of use of the pwm.
I took a look at your library and I think you log in directly if you have registers not using the standard arduino spi library.
I think it is enough to replace the names with the spi registers.
If all the spi configuration is in the "u8g_com_atmega_hw_spi.c", "u8g_com_arduino_hw_spi.c" files it shouldn't be too difficult, I could try.
Or I have to change my hardware and add chips.

olikraus, 2019-08-24:
Oh, this my old lib, yes, you could update the code...

tistructor, 2019-08-25:
thank you,
I evaluate what is best to do.