olikraus/u8glib #498: Having a problem display SD text file onto LCD using u8glib

TheDIYGuy, 2019-09-28:
Hello, i seem to be having a problem displaying a sample text file onto my lcd screen. I have the reprap Fullgraphics smart controller.

The lcd is a 128 x 64
It also has an sd card.
It took awhile but eventually I got the lcd working and the sd card working at the same time. As in, I can read the text file and see it on the serial com, and also upload a graphic to the lcd. The problem i face is that i cant display the contents of the text file on the lcd. Its a simple 5 letter text file (You Yo). When i try to display the characters of the text file, I get the acsii numbers. I want to be able to display the characters, You Yo, which are in the text file, onto my lcd. Any help would be greatly appreciated

My code:

#include "U8glib.h"  // u8

U8GLIB_ST7920_128X64_1X u8g(6, 5, 9); // u8

#include <SPI.h>   //sd
#include <SD.h>    //sd

const int chipSelect = 4;  //sd 
int i = 0;

File dataFile;
String text;
int text1 = 1;
void setup() {// setup is all sd
   // Open serial communications and wait for port to open:
   
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  Serial.print("Initializing SD card...");

  // see if the card is present and can be initialized:
  if (!SD.begin(chipSelect)) {
    Serial.println("Card failed, or not present");
    // don't do anything more:
    while (1);
  }
  Serial.println("card initialized.");

  // open the file. note that only one file can be open at a time,
  // so you have to close this one before opening another.
  File dataFile = SD.open("test.txt");

  // if the file is available, write to it:
  if (dataFile) {
    
    /*while (dataFile.available()) {
      Serial.write(dataFile.read()); 
      
    }
    dataFile.close();
    */
    while (dataFile.available()) {
      char text[3];
      
      
     text[i] = dataFile.read();
     Serial.print(text[i]); 
    
     i++; 
    // Serial.print(i);
       
    }
    dataFile.close();
  }
  // if the file isn't open, pop up an error:
  else {
    Serial.println("error opening datalog.txt");
  }
  
 
     
}

void loop() {// loop is all u8
 
u8g.firstPage(); 
 do {
  Draw();
  } while( u8g.nextPage() );
  delay(200); 
  text1++;

  }

  void Draw(){
u8g.setFont(u8g_font_gdr25r);
//u8g.drawStr(8, 30, "Eeee");

char buf[50];
sprintf(buf, "%d", text);
u8g.setPrintPos(40,60);
u8g.print(buf);
//u8g.drawStr(0, 20, );
//u8g.drawStr(40, 20, buf);
u8g.drawStr(0, 60, "JP");

    }

TheDIYGuy, 2019-09-28:
Solved!

I had text defined as a string in my previous code, all i had to do was change it to be a char and do u8g.print(text)     or
u8g.drawStr(40,20,text)

Didnt even need the buffer after all

New code

/*
#include "U8glib.h"  // u8

U8GLIB_ST7920_128X64_1X u8g(6, 5, 9); // u8

#include <SPI.h>   //sd
#include <SD.h>    //sd

const int chipSelect = 4;  //sd 
int i = 0;

File dataFile;
char text[3];
int text1 = 1;
void setup() {// setup is all sd
   // Open serial communications and wait for port to open:
   
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  Serial.print("Initializing SD card...");

  // see if the card is present and can be initialized:
  if (!SD.begin(chipSelect)) {
    Serial.println("Card failed, or not present");
    // don't do anything more:
    while (1);
  }
  Serial.println("card initialized.");

  // open the file. note that only one file can be open at a time,
  // so you have to close this one before opening another.
  File dataFile = SD.open("test.txt");

  // if the file is available, write to it:
  if (dataFile) {
    
    /*while (dataFile.available()) {
      Serial.write(dataFile.read()); 
      
    }
    dataFile.close();
    */
    while (dataFile.available()) {
      
      
      
     text[i] = dataFile.read();
     Serial.print(text[i]); 
    
     i++; 
    // Serial.print(i);
       
    }
    dataFile.close();
  }
  // if the file isn't open, pop up an error:
  else {
    Serial.println("error opening datalog.txt");
  }
  
 
     
}

void loop() {// loop is all u8
 
 char buf[50];
sprintf(buf, "%d", text);

Serial.print(i);
i++;
u8g.firstPage(); 
 do {
  Draw();
  } while( u8g.nextPage() );
  delay(200); 
  text1++;

  }

  void Draw(){
u8g.setFont(u8g_font_gdr25r);
//u8g.drawStr(8, 30, "Eeee");

char buf[50];
sprintf(buf, "%d", text);
u8g.setPrintPos(40,60);
u8g.print(text);
//u8g.drawStr(0, 20, );
u8g.drawStr(40, 20, text);
u8g.drawStr(0, 60, "JP");

    }

*/

TheCook2, 2019-10-07:
This is the closest I have come in the last few days to accessing an SD card and my LCD.  Outstanding everyone!  I am using a UNO and am a little pin challenged at the moment in that I don't have enough spare pins to build a second SPI.

I tried this with "U8GLIB_ST7920_128X64_1X u8g(13,11, 9,8);" in an attempt to use the same SPI SCK and MOSI as the SD card rather than make a complete second port.  It doesn't work.  Does this mean that SPI is a port rather than a bus.  Do I have to build a new SPI port for each device?  If so I think I will have to chuck my display and go for an i2c device.  Anyone seen an SD card reader that uses the i2c bus?

I suppose my real question is... Is it possible to use a shared SPI bus with SD devices and a u8g LCD?  

I have tried using setHardwareBackup but it has no effect.  I still get nothing on my LCD.  I did note that the example in the user reference does use a second SPI "U8GLIB_DOGM128 u8g(7, 5, 1, 2);"  Would it work if I bought a  SD card device to use this (7,5,1,2) SPI port?  (I am also financially challenged in that I can't go off buying bits of hardware that don't work eg a128x64 LCD display that doesn't play with other devices on the standard SPI interface.  Having said that, it looks like I can skip a walk to the cafe and buy an SD device for a dollar or two if I wait two months for the slow boat from China).  Noting that pin one is used for serial comms and may be problematic.

TheCook2, 2019-10-07:
Ahhhhh... tried it with "U8GLIB_ST7920_128X64_1X u8g(9,8);" and it looks like it is working.

Pulling head in.

olikraus, 2019-10-07:
Please note that I do not support u8glib any more. U8g2 is the more recent version.