olikraus/u8glib #499: Mirror Display 128x32 "0.96 ?

MagierKummer537, 2019-10-19:
Hi olikraus,
I really enjoy your u8g Libary. Good work and sorry for my bad english. But i wanna mirror/invert text of my 128x32 SSD1306 "0.96 Display, wich I defined in my Code with "U8GLIB_SSD1306_128X32 u8g(U8G_I2C_OPT_NONE);". I know, it has to do with this 0x0c8, but i dont really know, what to do...
Hope you can help me. :D
Thank you!

MagierKummer537, 2019-10-19:
Ich merke gerade, du bist ja Deutscher. :)
Dann mach dir keine Mühe mit der Übersetzung.
Ich möchte mein genanntes Display spiegeln, bzw. dessen Text gespiegelt rausgeben lassen, ich muss ja irgendwas in der Display- Datei ändern, nur was? ist das in der 2. Version der Libary einfacher?
Danke dir im Vorraus.

MagierKummer537, 2019-10-19:
Okay, mit der u8g2 Lib kann ich es entweder spiegeln (U8g2_MIRROR) oder die Rotation auf 270 Grad stellen (..._R3), ABER nicht beides. Ich brauche beides. Wie geht das denn nun?

olikraus, 2019-10-20:
> Wie geht das denn nun?

Leider derzeit gar nicht. Du willst tatsächlich hochkannt (270 Grad) und gespiegelt?
Also, für den SSD1306 müsste folgendes gehen: R3 verwenden und den init wert von c0 auf c8 manuell ändern:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1306_128x64_noname.c#L56

MagierKummer537, 2019-10-22:
Ok, danke für die Antwort. Sorry fürs späte Lesen. Ich probiere das mal. :D

MagierKummer537, 2019-10-22:
Das ist doch aber der 128x64 nicht mein 128x32, ist das trotzdem die selbe Datei.

MagierKummer537, 2019-10-22:
Habe das mal in der betreffenden Datei (SSD1306 128x32) geändert, es wird zwar um 270 Grad gedreht, aber nicht gespiegelt.

olikraus, 2019-10-22:
> Das ist doch aber der 128x64 nicht mein 128x32, ist das trotzdem die selbe Datei.

Welcher construktor ist es denn in der U8g2 (oben hattest du den U8glib construktor angegeben)?