olikraus/u8glib #500: Incompatibility with Arduino NANO EVERY board

michelfrance78, 2019-10-31:
Hello,
The U8glib library is incompatible with the new Arduino NANO EVERY board. Is an update planned for this likely replacement board of the basic NANO ?

olikraus, 2019-11-01:
Compatibility was one reason why I started u8g2.
Can you check the newer u8g2 lib?

michelfrance78, 2019-11-01:
> Message du 01/11/19 07:25
> A : "olikraus/u8glib"
> Copie à : "michelfrance78" , "Author"

Thank you for your fast reply. I did not know this new library. I was able to migrate all my code from a basic Nano card to a Nano Every card. I only noticed that compared to the U8glib library, the scrolling texts are about 4 to 5 times slower.
Cordially.

olikraus, 2019-11-01:
You mean u8g2 is slower?

michelfrance78, 2019-11-02:
> Message du 01/11/19 23:21

Yes I could compare on the vertical scrolling of 3 texts between a configuration Nano basic rev. 3 with U8g vs Nano Every with U8g2. The speed of scrolling is of the order of 4 to 5 times slower in the second case. Here are the extracts of the codes (I2C configuration) :

1. Nano basic + U8glib :

#include  
#include "U8glib.h"
...
U8GLIB_SSD1306_128X64 u8g(U8G_I2C_OPT_NO_ACK);
...
for (int a = 0; a < 20; a++)
{
u8g.firstPage();
do
{
u8g.setFont(u8g_font_fub11);
u8g.setFontRefHeightExtendedText();
u8g.setDefaultForegroundColor();
u8g.setFontPosTop();
u8g.setPrintPos(2, a);
u8g.print(F("TEXTE1"));
u8g.setPrintPos(2, a+16);
u8g.print(F("TEXTE2"));
u8g.setPrintPos(2, a+32);
u8g.print(F("TEXTE3"));
}
while (u8g.nextPage());
}
delay(3000);
...

2. Nano Every + U8g2 :

#include 
#include 

//#ifdef U8X8_HAVE_HW_SPI
#include 
//#endif
//#ifdef U8X8_HAVE_HW_I2C
#include 
//#endif
...
U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE); // All Boards without Reset of the Display
...
u8g2.clearBuffer();
u8g2.setFont(u8g2_font_fub11_tf);
for (int a = 0; a < 20; a++)
{
u8g2.clearBuffer();
u8g2.setCursor(2, 11+a);
u8g2.print("TEXTE1");
u8g2.setCursor(2, 27+a);
u8g2.print("TEXTE2");
u8g2.setCursor(2, 45+a);
u8g2.print("TEXTE3");
u8g2.sendBuffer(); 
}
delay(3000);
...

For other functions I did not compare because the speed is perfect for my projects.
For information, here are also the results of the small performance test FPS given in the examples of the U8g2 library :
- line 1.9 (Nano Every) 2.0 (Nano basic)
- clear screen : 1.9 (Nano Every) 2.1 (Nano basic)
- clip test : 1.9 (Nano Every) 2 
- draw @ : 1.6 (Nano Every) 1.7 (Nano basic)
- draw pixel : 1.7 (Nano Every) 1.8 (Nano basic)
 (all these tests exclusively use the U8g2 library)

michelfrance78, 2019-11-03:
Indeed when I use HW_I2C I get a scrolling speed probably even faster than my old configuration Nano Basic + U8Glib.
I consider my question totally resolved. My project is now fully migrated to a Nano Every + U8g2lib configuration. I thank you for your prompt assistance. Sincerely.

closed 2019-11-03 by michelfrance78