olikraus/u8glib #503: Nano 33 BLE support

derekchall, 2019-12-09:
I get the following when trying to compile for the new Nano33 BLE board, will this board be supported?

Thanks

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\derek\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\derek\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\derek\Documents\Arduino\libraries -fqbn=arduino:mbed:nano33ble -vid-pid=0X2341_0X805A -ide-version=10810 -build-path C:\Users\derek\AppData\Local\Temp\arduino_build_137704 -warnings=default -build-cache C:\Users\derek\AppData\Local\Temp\arduino_cache_235334 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.openocd.path=C:\Users\derek\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.openocd-0.10.0-arduino9.path=C:\Users\derek\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\derek\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\derek\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.bossac.path=C:\Users\derek\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\derek\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -verbose C:\Users\derek\AppData\Local\Temp\arduino_modified_sketch_866373\GraphicsTest.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\derek\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\derek\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\derek\Documents\Arduino\libraries -fqbn=arduino:mbed:nano33ble -vid-pid=0X2341_0X805A -ide-version=10810 -build-path C:\Users\derek\AppData\Local\Temp\arduino_build_137704 -warnings=default -build-cache C:\Users\derek\AppData\Local\Temp\arduino_cache_235334 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.openocd.path=C:\Users\derek\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.openocd-0.10.0-arduino9.path=C:\Users\derek\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino9 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\derek\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\derek\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.bossac.path=C:\Users\derek\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -prefs=runtime.tools.bossac-1.9.1-arduino1.path=C:\Users\derek\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino1 -verbose C:\Users\derek\AppData\Local\Temp\arduino_modified_sketch_866373\GraphicsTest.ino
Using board 'nano33ble' from platform in folder: C:\Users\derek\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.3
Using core 'arduino' from platform in folder: C:\Users\derek\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.3
Detecting libraries used...
"C:\\Users\\derek\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g -Os -nostdlib "@C:\\Users\\derek\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.3\\variants\\ARDUINO_NANO33BLE/defines.txt" "@C:\\Users\\derek\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.3\\variants\\ARDUINO_NANO33BLE/cxxflags.txt" -DARDUINO_ARCH_NRF52840 -mcpu=cortex-m4 -w -x c++ -E -CC -DARDUINO=10810 -DARDUINO_ARDUINO_NANO33BLE -DARDUINO_ARCH_MBED "-IC:\\Users\\derek\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.3\\cores\\arduino" "-IC:\\Users\\derek\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.3\\variants\\ARDUINO_NANO33BLE" "-IC:\\Users\\derek\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.3\\cores\\arduino/api/deprecated" "-iprefixC:\\Users\\derek\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.3\\cores\\arduino" "@C:\\Users\\derek\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed\\1.1.3\\variants\\ARDUINO_NANO33BLE/includes.txt" "C:\\Users\\derek\\AppData\\Local\\Temp\\arduino_build_137704\\sketch\\GraphicsTest.ino.cpp" -o nul
Alternatives for U8x8lib.h: [U8g2@2.26.14 arduino_993763@2.25.10]
ResolveLibrary(U8x8lib.h)
  -> candidates: [U8g2@2.26.14 arduino_993763@2.25.10]
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x0 pc=0x7855fb]

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder.ResolveLibrary(0x11cce5a0, 0x11cf6337, 0x9, 0x11cf6337)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/resolve_library.go:64 +0x14b
github.com/arduino/arduino-cli/legacy/builder.findIncludesUntilDone(0x11cce5a0, 0x11dd4080, 0x88ed00, 0x11e067e0, 0x11c06750, 0x11c04100, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:358 +0x3b6
github.com/arduino/arduino-cli/legacy/builder.(*ContainerFindIncludes).Run(0xdebc68, 0x11cce5a0, 0xdebc68, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:152 +0x3e5
github.com/arduino/arduino-cli/legacy/builder.runCommands(0x11cce5a0, 0x11c5de1c, 0x20, 0x20, 0x409a01, 0x3fff, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:210 +0xbd
github.com/arduino/arduino-cli/legacy/builder.(*Builder).Run(0x11c69f40, 0x11cce5a0, 0xdebbb4, 0x11c06c98)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:117 +0xb9c
github.com/arduino/arduino-cli/legacy/builder.RunBuilder(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:226
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:398 +0x6d2
C:\Program Files (x86)\Arduino\arduino-builder returned 2
Error compiling for board Arduino Nano 33 BLE.

olikraus, 2019-12-10:
I do not support u8glib any more, what about u8g2?

derekchall, 2019-12-10:
arr looks like I am a bit out of date, u8G2 works perfectly thanks