olikraus/u8glib #504: There is no 160x160 display supported

mortaca, 2020-01-10:
Hello, I would like to use a 160x160 screen and since there is a closed issue to use the uc1698 driver because it is very difficult to implement, it is actually the most used in this particular resolution, in that thread it is said that it is better to use the uc1610 but in the list of supported screens the uc1610 is only available for a 160x104 screen and I have realized that there really is no support for any 160x160 screen.

Would it be possible to add some model? I leave to your choice which you think that is simpler and if to be a COG and that if it is through port 6800 it would be preferable also to have a lower communication lag, the idea is to use this display to play video games so the Soft drink is very important and the SPI port I think can be problematic.

Currently I have two screens one with the UC1698 and another with a UC1610 both 160x160, there is also another model with the RA8822 but the driver on a board is not so practical.

Thanks in advance for the incredible support this library has and the work you do.

olikraus, 2020-01-10:
I do not extend u8glib any more, but I could add support for a UC1610 160x160 display in u8g2lib.
There is already support for a uc1611 160x160 display in u8g2: https://github.com/olikraus/u8g2/issues/884

RA8822: I have never heard about this controller
UC1698: Actually i tried to add support for this display but I failed. It is also not a monochrome controller but instead a RGB controller, which is used for monochrome displays.

Anyhow, I think the UC1611 160x160 support could be added in u8g2.

mortaca, 2020-01-11:
Great! I found this display with UC1611 and 160x160;
€ 10,95 | LG1601601 160x160X62x69mm gráfico lcd COG de UC1611 SPI CII I2C interfaz paralela, 3,3 v de caracteres lcd FSTN negro de alta calidad gran oferta
https://a.aliexpress.com/1YRizhtbH
This can be valid, what do you thing about the comunication protocol and the refresh rate? The 6800 maybe can be better to get less input lag compared with the SPI serialized bus?

Many thanks for the fast reply.

olikraus, 2020-01-11:
SPI will be faster if you use the hardware SPI support of your board. 
Refresh rate will depend on the amount of data on the display and your microcontroller. I personally would suggest Arduino MKR variants. ESP32 or the new Seeedstudio XIAO would be fast enough.

mortaca, 2020-01-11:
Yes, specifically we are working with ESP32.
Great, we will be looking forward to the update and thank you very much for your work.