olikraus/u8glib #506: SBN1661 with R/W but no Reset

Maes2ro, 2020-05-19:
I am working with a 122x32 graphics display that uses the SBN1661 chip. (please see attached data sheet)
I have written my code using the U8G2_SBN1661_122X32_F Constructor, however the Midas display does not have a Reset line, but it does have an R/W line. At the moment, I have tied the R/W low (tied to W) and have placed U8X8_PIN_NONE in place of a Reset pin.

As I cannot get a proper behaviour from my display, I think have either made an error with my board, or have not configured the constructor properly. My code is just your 'hello world' example

I would appreciate any guidance you might be able to offer on setting up the constructor. 

[MIDAS 122x32 transflective display.pdf](https://github.com/olikraus/u8glib/files/4652649/MIDAS.122x32.transflective.display.pdf)

olikraus, 2020-05-19:
I think this issue should be moved to u8g2 project, right?

What exactly do you see on the screen?

Maes2ro, 2020-05-25:
Sorry for the delayed reply.

In order to take away the variable of my board with level shifters. I have
now built a test assembly using an Arduino Uno that already has signal
levels at 5V.

This is the code I am running
/* Midas display test
 *
 */
//************************** include header
files*******************************
 #include <U8g2lib.h>
//********************** variable declarations
***********************************

//*************************
constructors***************************************
U8G2_SBN1661_122X32_F u8g2(U8G2_R0, 3, 4, 5, 6, 7, 8, 9, 10, 0, 1, 2, 11);
/*U8G2_SBN1661_122X32_F(rotation, d0, d1, d2, d3, d4, d5, d6, d7, dc, e1,
e2, reset) [page buffer, size = 256 bytes]*/
//**********************definitions
***********************************************
#define A0 0
#define E1 1
#define E2 2
#define DB0 3
#define DB1 4
#define DB2 5
#define DB3 6
#define DB4 7
#define DB5 8
#define DB6 9
#define DB7 10
#define R_W 11

void setup() {
  u8g2.begin();
 }

void loop() {
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_ncenB14_tr);
  u8g2.drawStr(0,20,"Hello World!");
  u8g2.sendBuffer();
}

As you can see I connected the R/W line to pin 11 (where Reset is in the
constructor)

The basic image I see is a rectangular shape - I guess all pixels ON. - no
text at all
With R/W connected to pin 11 (reset line), then the grey box flickers -
probably at the scan rate of the Arduino.
With R/W connected to GND (display input pinned to W) then the displayed
box is steady, with no flickering.

Any help with diagnostics would be extremely gratefully received.

Maes2ro, 2020-05-25:
To see if I was missing anything obvious, I returned to your examples, and
noticed that I was not explicitly including <Arduino.h>. I made this
change, and the display then displayed Hello World! perfectly.

This seemed a bit strange, as the compiler was not flagging an error
before, and I thought that arduino.h was part of the IDE anyway?. To try
and cross-check this, I then commented out the <Arduino.h> include, but the
programme still ran fine.

I am putting  U8X8_PIN_NONE for the reset line and tying R/W low in my
hardware.

To run another test, I have just used my constructor in your graphics
example, and that also works perfectly. Now I know that my example can run
correctly, I will return to my original board with the level shifters. to
check the hardware.

Thank you very much for your help so far, and I am sure that I will have
more questions later, when I have to port the whole thing across to
Raspberry Pi.

Maes2ro, 2020-06-04:
This is working well now, thank you

closed 2020-06-04 by Maes2ro

olikraus, 2020-06-05:
:-)