olikraus/u8glib #509: kurze Frage / short Question

Greenislands, 2020-06-29:
Grüß dich Oli,

I am a greenhorn in Arduino and programming. Hours of hard testing I then finally got it. A teensy 3.2 board runs the "Hello World" sketch on a Nokia 5110 display. Now to the point. The task is to join two functions: To draw with u8glib to the display data from another library with getting those data with a function.

function of the other library ist this:
`void onIasUsChange(char* newValue) {
    /* your code here */
}
DcsBios::StringBuffer<4> iasUsBuffer(0x0426, onIasUsChange);`

So here is the conflict. The u8glib starts drawing by:
`void draw(void) {`

I am not able to integrate the two functions to work with. My first intention dont work:

`U8GLIB_PCD8544 u8g(13, 11, 10, 9, 8);    // SPI Com: SCK = 13, MOSI = 11, CS = 10, A0 = 9, Reset = 8

void onIasUsChange(char* newValue) {
  // graphic commands to redraw the complete screen should be placed here  
  u8g.setFont(u8g_font_unifont);
  //u8g.setFont(u8g_font_osb21);
  u8g.drawStr( 0, 22, "newValue");

}
DcsBios::StringBuffer<4> iasUsBuffer(0x0426, onIasUsChange);

void setup(void) {
  // flip screen, if required
  // u8g.setRot180();
  
  // set SPI backup if required
  //u8g.setHardwareBackup(u8g_backup_avr_spi);

  // assign default color value
  if ( u8g.getMode() == U8G_MODE_R3G3B2 ) {
    u8g.setColorIndex(255);     // white
  }
  else if ( u8g.getMode() == U8G_MODE_GRAY2BIT ) {
    u8g.setColorIndex(3);         // max intensity
  }
  else if ( u8g.getMode() == U8G_MODE_BW ) {
    u8g.setColorIndex(1);         // pixel on
  }
  else if ( u8g.getMode() == U8G_MODE_HICOLOR ) {
    u8g.setHiColorByRGB(255,255,255);
  }
  
  pinMode(8, OUTPUT);
}

void loop(void) {
    DcsBios::loop();
  // picture loop
  u8g.firstPage();
  do {
    onIasUsChange;
  } while (u8g.nextPage());
  

  
  // rebuild the picture after some delay
  //delay(50);
}`

Can anyone support with his magic?
thanks in advance

Maik

olikraus, 2020-07-01:
Hallo Maik

There are multiple ways to solve this problem. Here is one suggestion:

1. Introduce a global buffer: char msg[32];
2. Fill that buffer in the onIasUsChange function with strncpy
3. In the main loop, output the value of "msg"

``` C++
char msg[32];

void onIasUsChange(char* newValue) {
  strncpy(msg, newValue, 32);
 msg[31] = '\0';
}
```

Then later, do something like this:
``` C++
void loop(void) {
    DcsBios::loop();
  // picture loop
  u8g.setFont(u8g_font_unifont);
  u8g.firstPage();
  do {
   u8g2.drawStr(0, 22, msg);
  } while (u8g.nextPage());
}
```

Greenislands, 2020-07-01:
Hey Oli,

it works exactly the way you showed!
That kicks my project of building an arduino based cockpit forward.

**Big thanks** to you for sharing your knowledge! I owe you some beer

Greenislands, 2020-07-01:
Okay Oli,

the programming got me another time.
```
void onAltMslFtChange(unsigned int* newValue) {
  strncpy(ALT, newValue, 32);
  ALT[31] = '\0';
}
DcsBios::IntegerBuffer altMslFtBuffer(0x0432, 0xffff, 0, onAltMslFtChange);
```
Okay, this time I got another datacall which seems to be an integer. I am looking if there is any other surprise datatype. no... "unsigned int" is common, "char" is rare.
What I have found out is that I cant use strncpy for copying unsigned int values.

olikraus, 2020-07-02:
The msg[32] only was a example. You can use any data.

``` C++
unsigned int n;
void onAltMslFtChange(unsigned int* newValue) {
  n = newValue;
}
```

Then later, do something like this:

``` C++
void loop(void) {
    DcsBios::loop();
  // picture loop
  u8g.setFont(u8g_font_unifont);
  u8g.firstPage();
  do {
    DcsBios::loop();
   u8g2.setCursorPos(0,22);
   u8g2.print(x);
  } while (u8g.nextPage());
}
```

Greenislands, 2020-07-03:
hey ho,

I got it,

`setCursorPos`

don´t work. After studying the u8g wiki I used `setPrintPos` instead.

Thank you oli. I swear I read the fucking manual first next time, but I feared of having no basic knowledge of c programming.

olikraus, 2020-07-04:
> don´t work. After studying the u8g wiki I used setPrintPos instead.

Maybe my mistake. I do not support u8g any more, instead u8g2 is out since several years and I guess I changed the function name here..