olikraus/u8glib #516: u8glib with PIC Microcontrollers

ahmedbasheer4, 2021-06-04:
Is there anyone who can run u8glib with any family of PIC MicroControllers?

olikraus, 2021-06-06:
U8glib and u8g2lib will not run on any PIC controller unless you have the professional version of the PIC C Compiler.
The freeware version of the PIC C Compiler does not include a linker garbage collector which is a required precondition for u8glib and u8g2lib. The linker garbage collector is part of the GNU compiler, so u8glib/u8g2lib will run on any system which is supported by GNU C Compiler.

Reference: https://www.microchip.com/forums/m668638.aspx

closed 2021-06-06 by olikraus

ahmedbasheer4, 2021-06-06:
thanks a lot
I succeeded to run the library on atmega32 and atmega2560 on Atmel studio and I will try to run it on PIC18F452.