olikraus/u8glib #517: JFrog shut down?

anotherboringuser, 2021-07-09:
Hi there, 
I just tried clicking on a link in readme.md here on GitHub to download an older version of u8glib and I ended up at this website: https://jfrog.com/center-sunset/ 
What did I miss? 
Is there a way to get u8glib 1.17?

olikraus, 2021-07-09:
1.17?
Was this ever published? https://github.com/olikraus/U8glib_Arduino/releases
https://www.arduinolibraries.info/libraries/u8glib

anotherboringuser, 2021-07-09:
I think there was a version 1.17, at least according to this guy: https://youtu.be/lAKyZd63_ns?t=553 

But what about those links?

olikraus, 2021-07-10:
Hmm, I once used a service to store u8glib releases. I do not remember that service name, but maybe this got acquired by jfrog.
Anyhow, I tried to reconstruct 1.17 zip file from the repo, which is now stored here: https://github.com/olikraus/u8glib/tree/master/download

It is called 1.17pre6 but actually this should be v1.17.

asklvbuimn, 2021-08-03:
Hi

i found the same problem with jfrog but after some google searched, i found this link.
https://www.14core.com/wiring-the-128x64-1-3-inch-oled-display-on-4-wire-3-wire-i2c-interface/u8glib_arduino_version1-17/
after that only i saw this comment and got the file here.
just wanna confirm on this file as it seems a bit larger