olikraus/u8glib #521: How do i connect my KS0108?

jansaternus, 2022-03-27:
I want to connect my KS0108 LCD Display to arduino uno but i can't fint the right one. Please help he because i spent 2 hours trying to find a good image of how to connect it. If someone can decode the numbers in the constructor then i maybe can do it.

olikraus, 2022-03-27:
https://github.com/olikraus/u8g2/issues/1823

olikraus, 2022-03-27:
Connection is very similar to this tutorial: https://docs.arduino.cc/learn/electronics/lcd-displays
More u8g2 specific: A tutorial is here: https://github.com/olikraus/u8g2/wiki/setup_tutorial#identify-digital-lines
And my most loved video on this topic is here: https://www.youtube.com/watch?v=4CD8ERaylmY

closed 2022-12-24 by jansaternus