olikraus/u8glib #526: 16lvl Grayscale Bitmap?

Jaboop8, 2023-02-03:
Was looking at the documentation for this library and wasn't seeing an option for a 16lvl grayscale bitmap. Is there a function that can do this? I am currently using the drawXBMP for a monochrome bitmap but was hoping to make a nicer grayscale image. Is this achievable?

olikraus, 2023-02-04:
U8glib is not extended any more. I think such a function is not there

Jaboop8, 2023-02-08:
@olikraus thanks for confirming. There isn't a grayscale u8glib2 library yet right? Are there any grayscale OLED drivers that you've seen that are active?

olikraus, 2023-02-19:
Since many years i wanted to write a grayscale lib, but it never happened :-(
And no: I do not know any grayscale lib.