olikraus/u8glib #527: Wrong render

jansaternus, 2023-02-14:
When i connect my KS0108 to my Arduino Due it renders some letters wrong. 0 and 1 are rendered just fine but 2 and 3 are not, and instead of "rotary" it prints "rötäry"

olikraus, 2023-02-14:
Strange. Any code / pictures?

jansaternus, 2023-02-15:
[image]
 Here is one

olikraus, 2023-02-15:
Thanks. Can you also post the code which causes this?

olikraus, 2023-02-19:
You must not change the content of the picture loop itself. This means, the millis() check should be outside the loop.
More like this (not tested):
``` C++
  } 
  if(millis() - lastChange > 2000){
    u8g.firstPage();
    do{
        for(int i = 0; i < SAMPLES/2; i++)
          u8g.drawLine(i*2, 64, i*2, 64-fft[i]);
    }while(u8g.nextPage());
  } else{
    u8g.firstPage();
      u8g.setFont(u8g_font_unifont);
      u8g.setPrintPos(0, 20); 
      u8g.print("Rotary:");
      u8g.setPrintPos(0, 40); 
      u8g.print("Count");
     }while(u8g.nextPage());
  }

```

jansaternus, 2023-02-21:
Well it dosen't display the text but the FFT yes