olikraus/u8glib #530: U8g2 used PROGMEM AtmelStudio7

NeiD52, 2023-08-06:
Hello! I'm trying to use the U8g2 library to output a string from an array, I took into account the recommendations that were already given on the GitHub, but I can't understand how this is done correctly and whether it is necessary to apply strcpy_P
My code, and no text on display
````const char* const messages[] PROGMEM =
{
"first",
"second",
....
};

strcpy_P(lcd_buf, messages[0]);
u8g2_DrawUTF8(&u8g2, 2, 7, messages[0]); 
_**or**_
strcpy_P(lcd_buf, messages[0]);
u8g2_DrawStr(&u8g2, 2, 7, messages[0]); 
```

Only works like this
`u8g2_DrawUTF8(&u8g2, 2, 7, messages[0]);`

Tried to do so, but without success
```
strcpy_P(lcd_buf, (const char*)pgm_read_word(&messages[0]));
u8g2_DrawUTF8(&u8g2, 2, 7, messages[0]);
```

olikraus, 2023-08-06:
Use lcd_buf as argument for u8g2 functions.

NeiD52, 2023-08-06:
> Use lcd_buf as argument for u8g2 functions.
Is this a suggestion or a question? give an example code

NeiD52, 2023-08-06:
> Use lcd_buf as argument for u8g2 functions.

is this a suggestion or a question? give an example code

olikraus, 2023-08-06:
I use my smartphone at the moment, so typing is limited. Use
u8g2_DrawUTF8(&u8g2, 2, 7, lcd_buf);

NeiD52, 2023-08-06:
> I use my smartphone at the moment, so typing is limited. Use u8g2_DrawUTF8(&u8g2, 2, 7, lcd_buf);

should I not use strcpy_P?
if you are at the computer, please give a full answer?

olikraus, 2023-08-06:
No, just replace that one line in your code

NeiD52, 2023-08-06:
> No, just replace that one line in your code

strcpy_P(lcd_buf, messages[0]);
u8g2_DrawUTF8(&u8g2, 2, 7, lcd_buf);

it was done, I made a mistake and posted that I use messages[0] instead of lcd_buf, but it doesn't work

olikraus, 2023-08-06:
The code in your last comment should be correct, provided, that the definition of the array is correct. 
In fact your question is a little bit out of the scope for this library. I am not an expert on PROGMEM topics and it actually has nothing todo with u8g2 (and BTW you have used the old issue list from u8glib, which is the predecessor of u8g2).

Another alternative way might be to use https://github.com/olikraus/u8g2/wiki/u8g2reference#print instead of DrawUTF8
The print function is derived from Arduino Framework and might be more flexible regarding PROGMEM.

NeiD52, 2023-08-07:
> The code in your last comment should be correct, provided, that the definition of the array is correct. In fact your question is a little bit out of the scope for this library. I am not an expert on [...]

Dear olikraus!  You were right!  as soon as I correctly defined the array and the arrays of references to it, text appeared on the display, and the RAM memory decreased by 25%.  Thank you for your help!  I gave an example of my corrections in the code.
```
const char string_0[] PROGMEM = "String 0"; // "String 0" etc are strings to store - change to suit.
const char string_1[] PROGMEM = "String 1";
const char string_2[] PROGMEM = "String 2";
const char string_3[] PROGMEM = "String 3";
const char string_4[] PROGMEM = "String 4";
const char string_5[] PROGMEM = "String 5";

const char *const string_table[] PROGMEM = {string_0, string_1, string_2, string_3, string_4, string_5};
```

closed 2023-08-07 by NeiD52

olikraus, 2023-08-07:
Looks good