olikraus/u8glib #536: 1.3inch OLED Module (C)

Gerd4711, 2024-10-18:
I added the Library U8g2 to the Arduino R3 platform and tried to run the integrated example "Hello World".
The constructor line for "SH1107 - 128x64 - SPI" is released. The 3 mentioned wires are connected
as descibt in the constructor line. Further wires are connected to CLK*PIN11 and DIN*13.
What is the reason that the screen is still black!

olikraus, 2024-10-18:
> What is the reason that the screen is still black!

You provided far too less information to say something.

What is the exact constructor you used?
How are Arduino and Display wired (complete diagram including power)
Also send a picture...

Gerd4711, 2024-10-18:
This is the constructor line 

U8X8_SH1107_64X128_4W_HW_SPI u8x8(/* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);  CLK PIN11 and DIN PIN13.
[image]
[image]

It is my first time that I operate on this platform